ShikOfTheRa/scarab-osd #603: Display armed timer

sevet, 2020-04-24:
Sorry for the multiple requests.
Can armed timer be added or option to change "on time" timer to armed time.
The armed timer can have an option of total armed time or reset to zero every arm. (i prefer total)

ShikOfTheRa, 2020-04-24:
Current operation is:
Single dual purpose field - 
When armed it shows total "armed" time.
When disarmed it shows total "on" time (you can view "armed" time in statistics menu if needed.

memory is tight so easiest way for mem to suppor thi is as follows:
config.h option to chooses:
 1 - Dual purpose as it is now (default)
 2 - Only ever show "total" time
 3 - only ever show "armed" time
 4 - show both with second on line directly underneath. GUI will only be able to position the pair and turn both on or off. 

Would that work for you?

ShikOfTheRa, 2020-04-24:
Another option would we to re-purpose the remaining time timer.
You could then have two fully adjustable independent timers.

Would have to enable in config.h, but I think a good option...

sevet, 2020-04-24:
Sounds good to me, thanks.

sevet, 2020-04-24:
Both approches are good, but maybe add to gui option as it will get
forgotten in the config.h considering most ppl probably use the precompiled
versions now.

ShikOfTheRa, 2020-04-27:
config.h option #define DUALTIMER Enables option to use separate Flight Timer and On Timers. 
This re-purposes the "Flight Time" remaining GUI configurator field. 
Default is for this to be disabled as it permits use of 3 timers - On time, flight time and remaining time.

For now it will be config.h option only, but may be able to implement into the GUI at a later stage – but it depends upon memory available. It is really tight..

closed 2020-05-06 by ShikOfTheRa