ShikOfTheRa/scarab-osd #610: All values except battery sometimes freeze shortly after boot

hagfelsh, 2020-05-06:
Aeromax hardware, 
scarab-osd 1.9.1.5 configured with MWOSD configurator 1.9.11

Connection details:
pixhawk serial4 @ 57kbps, mavlink v1 protocol selected

Also consuming telemetry data:
serial 1: 3DR radio
serial 2: Frsky smartport telemetry via craft and theory conversion cable

Running arduplane 4.0.5

Problem statement:
Random chance, about 75% of the time--about 2 seconds after OSD begins reporting Mavlink data (mode can be seen as INIT), all but battery info, screen switching (assigned to channel 5), and flight mode will freeze in whatever position it was in and stay that way until I power cycle the entire plane.

About 50% of the time I can restore expected functionality if I connect my groundstation (wirelessly).  It's as though the traffic between the radio on serial1 influences what's on serial4.

Can you think of what's causing this?  Is there a way to reboot the OSD while it's powered on?  Perhaps a power-up sequence is causing a race condition?

ShikOfTheRa, 2020-05-06:
Not heard of that one. 
As some of those items are running it sounds like the OSD itself is not locked up as it continually loops and updates the data on the screen. 

Most probably the mavlink data is not being recieved / disabled somehow.

On mavlink tab, try disabling "autoconfigure serial port"

hagfelsh, 2020-05-06:
Will do! I'll report the results once I've done that.  What is autoconfigure changing?  Similarly, perhaps disconnecting TX from the OSD would have an effect.. or is that a requirement for the OSD to operate?

ShikOfTheRa, 2020-05-06:
Disconnecting TX does the same. It is not essential. Only used for auto configure
Auto configure attempts to configure ardupilot to send the mavlink data the OSD needs. If set up manually it is not needed at all.

I hope it makes a difference!

ShikOfTheRa, 2020-05-07:
The other thought...
The OSD Mavlink i.d is set as 99 (hardcoded). If the 3DR radio is set the same that would be unlucky. One of the two would have to be changed.

hagfelsh, 2020-05-07:
Oh that's interesting... do you know now to change a mavlink id on the 3DR?

ShikOfTheRa, 2020-05-08:
No idea. See how the other things work first. Much mor elikely to be the issue

hagfelsh, 2020-05-09:
Disconnecting the TX pin had no effect.  I got two flights in today before the wind was too much even for the skywalker.  First flight was perfect except airspeed remained stuck at 2mph the whole time.

2nd flight the whole overlay remained motionless except for mode, efficiency figures, & battery statistics.  Map mode wasn't responsive either.  I dug around in the SiK radio editing options on Mission Planner, but nothing screamed 'mavlink_id'.  There's what I imagine a radio id, but that wasn't set to 99, it was set to 25, which I understand is default, according to the documentation (https://ardupilot.org/copter/docs/common-configuring-a-telemetry-radio-using-mission-planner.html).

I'll see if I can find a way to influence the problem; I wonder if it happens when its the only telemetry unit?  Do you know if scarab-osd prefers MavLinkv1 or v2?  There's an option for that in the serial port type options.

hagfelsh, 2020-05-09:
Testing complete!  Nothing seems to influence the problem. I tried these combinations:

- OSD only

- OSD + FrSky (no 3DR radio)

- OSD + 3DR (no FrSky)

I rebooted three times under each config to make a weak attempt at stamping down randomness.  

I see some debugging utilities available via the configuration tool; would any of these be helpful in diagnosing the nature of the problem?

hagfelsh, 2020-05-09:
I just moved scarab-osd to serial2 in case the flow control feature was important.  The osd was on serial4.  this seems to have improved the chances that the problem does _not_ happen, which is promising.  Adding frsky to serial4 seems to have returned the problem more reliably.  This is interesting considering that the C&T cable is read-only....

Airspeed is still stuck at 2mph though.  The groundstation disagrees and reports changes as expected.

ShikOfTheRa, 2020-05-11:
Interesting. 

This reminds me of an issue with PX4 once before. Basically the flight controller was running so much stuff that it could not doe everything and throttled everything back and OSD was updating every few seconds only. It still worked, just slowly.

Airspeed is also odd.

I haven't seen anything like this at all!

Re diagnostic utilities - The default will say if telemetry pkts are being received and how many per second. It doesn't say which ones are received. I might have to write one to do this so can see what you are receiving.
Are you able to capture a tlog file so I can analyse it?

ShikOfTheRa, 2020-05-21:
HAve you had any luck resolving?

hagfelsh, 2020-05-23:
Sort of, but not as much luck as I'd hope!

I had to travel a bit the last week or so, but I was able to bring the plane.  With that deadline in mind, I tried a big change: I pushed minim-osd extra in.  Reliability has been considerably better, yet the same set of problems got me on one of my flights.

This entirely eliminates scarab-osd from the list of causes, leaving it to be a hardware defect (cables, terminals, aeromax), or an infrastructure config problem (eg: duplicate ids like you had mentioned).

If you'd like to keep hunting with me, I'd love the company!  But in all fairness, this doesn't appear to be [just] a scarab issue, unless scarab and the others share this problem fundamentally, somehow.

ShikOfTheRa, 2020-05-24:
Strange one this..

If you can send me a tlog from mission planner from when it fails I can replicate and at least see what info is coming from the FC.

lesykm, 2020-05-30:
Interestingly, I have similar issue.
In my case only mode (stab-althold-loiter) changes (Ch 5), so some connection between FC and OSD works.
And other data does not update from beginning, it's not kind of freeze (I checked this by turning on vehicle on non-leveled surface - AH was not updated from very beginning).
Other values - GPS coordinates, HUD, AH do not change.

OSD type: Micro Minimosd
OSD version: 1.9.1.5
FC: Pixracer 1.0
Arducopter version: 4.0.3
OSD connected to Telem 2@57k@Mavlink1.
Serial speed is same both in OSD and Mission Planner.

My layout attached:
[
[image]
](url)

lesykm, 2020-05-31:
OK, solved my problem, sorry for disturbing.
By unknown reason, SR2_* update rates were set to 0.
So, when I set them to 2Hz, everything works perfectly.

ShikOfTheRa, 2020-06-01:
@lesykm great you have solved. Ststems are so complex it is very easy to miss something like that.
It does seem a different issue to @hagfelsh

ShikOfTheRa, 2020-06-22:
I am closing this issue due to age. ANy issues, please re-open or comment on it when closed ..

closed 2020-06-22 by ShikOfTheRa

hagfelsh, 2020-06-26:
Bah! Sorry for the immense delay.  All of summer's activities took over and I more or less just stopped keeping up on things that live on a computer.  I have hours of flight on the minim-osd extra via the aeromax board and the problem persists just as described previously.

More information is available now, too: the flight summary screen frequently appears during flight, accompanied with a 'no flight data received' message.  That's a good clue.

Here's another fun twist: I pulled the feed off of serial one and duplicated it with a breadboard adapter.  The other consumer is the 3DR radio, which operates fine enough. I've never had great results with it unless the plane was flying pretty close to the ground station, but it seems to operate correctly. The problem still occurs just as it did previously with this connection method.

I've been flying out of range of my ground station lately so my tlogs are pretty torn up.  Would you be able to make use of dataflash logs? Or does it have to be a tlog since that's a reflection of the same feed the OSD is receiving?

ShikOfTheRa, 2020-06-26:
No worries...

tlog is really what I need. I am not that familiar with ardupilot to know if you can use dataflash etc. I will try to see...