ShikOfTheRa/scarab-osd #611: Cannot flash Ardupilot firmware to micro minimosd

Daddy-Boomalati, 2020-05-20:
I'm sorry to put the issue here, but I keep getting a ReCaptcha error on FPVLab.

I'm using the latest firmware and GUI builds and I keep getting an error. I can read and write to the board with whatever firmware came loaded on it, but I cannot flash it. Attached is the error message.

Thanks!
[image]

ShikOfTheRa, 2020-05-21:
FAQ # 1 Unable to flash and I get "Programming FAILED: sending 3020"

99% of time this is connectivity or power issue
Check you are selecting the correct comm port
TX of OSD must connect to RX of programmer
RX of OSD must connect to TX of programmer
Programmer requires all six pins connected
Nothing else must power the OSD when programming. Only the USB programmer

ShikOfTheRa, 2020-05-21:
https://github.com/ShikOfTheRa/scarab-osd/wiki/FAQ

Daddy-Boomalati, 2020-05-21:
Thank you!!!

Dave Smith

ShikOfTheRa, 2020-05-21:
Let me know if it works!

Daddy-Boomalati, 2020-05-21:
Yeah, I don’t know what is up with it. I know I have it wired correctly. All six pins in the correct order. I can read and write to it, just not upload firmware. It connects fine, but the bootloader is DOA or something.

I even tried a different computer in case it was that. I am going to readymaderc tomorrow, and I’ll see if one of their guys can flash it for me, or exchange it out.

Thank you so much for taking the time to respond. You’ve always been a stand-up guy.

Dave Smith

ShikOfTheRa, 2020-05-29:
Any luck?

The only things I think could impact this:

1 board is powered up by something else. Or have large capacitors on power line.
2 the DTR pin is not connected end to end.
3 faulty board / programmer. Maybe it is set to 3.3v instead of 5v?
4 the minim does not have standard Arduino 56k bootloader. I only know of one manufacturer who did this and it wasn’t a minim style board.
5 the bootloader is corrupt – but I would expect it not to start up so extremely unlikely.

Daddy-Boomalati, 2020-05-29:
I ordered a minim board off eBay and it worked fine. The only thing (in my small pool of knowledge) can think is that the micro board was corrupt. The minim board took right off, and actually fits my 3D-printed mount I made better anyway, so I’m going with it!

I really appreciate you taking the time to help me.

Thanks,
Dave

ShikOfTheRa, 2020-05-29:
Great. glad you are up and running!

closed 2020-06-01 by ShikOfTheRa