ShikOfTheRa/scarab-osd #616: betaflight4.2 compatibility

corrado-c, 2020-06-22:
hi, I realized that from version 4.2 I can't access the BF osd menu, only the mwosd menu run.
in previous versions both menus worked correctly. do I have to change anything in the BF CLI? or has BF lost any functions that allowed to use external mwosd?
thank you so much!

ShikOfTheRa, 2020-06-23:
I wasn't aware support for CMS had been stopped. I am surprised if that is true. 
I can test as soon as I have a betaflight setup working again.

corrado-c, 2020-06-23:
Thanks so much!!!  if I can do some tests let me know, I will gladly help you!  some verification, or other info.  just explain to me in a simple way how to do it :)

ShikOfTheRa, 2020-07-07:
Sorry for delay - had to build a quad to test...
I tried latest Betaflight - 4.2.0 with latest MWOSD and can confirm it worked OK for me. I used a Omnibus F4 for testing

Note:
OSD must be set to disabled in Betaflight GUI - configuration tab/other features

also, if you have ameneded the following in betaflight, set back to defaults:
osd_displayport_device = AUTO
displayport_msp_serial = 0

closed 2020-07-20 by ShikOfTheRa