ShikOfTheRa/scarab-osd #618: Direct current sensor reading problem

hulandy, 2020-09-05:
Hi there. I'm using a furling32 v1.6 esc for my kiss v2 rig, and a micro minimosd with mwosd firmware, but I can't get the direct current info from the esc, I tried unplug the tx wire from the esc as well. Just can't make direct current sensor working. And I tried to register at fpvlab, but failed. Please help.. just tell me how to setup this, I have tried both 1.1v and 5v def, neither of them worked.

ShikOfTheRa, 2020-09-06:
It is intended to get the current readings from the KISS FC  or alternatively a ESC that supports analaog. ALmost certinly OSD should be set to "Use FC" for Amps
I think the furlong does not support analog so i would guess the KISS FC reads telemetry from the ESC and then sends back to the OSD. 
I do no know if KISS supports furlong (blheli telemetry)

hulandy, 2020-09-06:
Yep. Blheli32 use ADC for current sensor output, Although I don't know what's that mean. Now I roughly realized that it just doesn't work with the osd which only accept analog signal. Thank you for your reply and your works. For furling32, its telemetry output the voltage and temperature, rpm, but no current information unfortunately.

closed 2020-09-06 by hulandy