ShikOfTheRa/scarab-osd #619: Mwosd current reading for kiss

hulandy, 2020-09-07:
I just found out that the osd actually accepted the current sensor signal from my esc(furling32 v1.6), when it connected to the computer and with the lipo on, the problem is the numbers doesn't move in my goggles, I've tried every parameter of it. Could you test it standalone? I think you don't need to connect it to a kiss FC. Thanks.

ShikOfTheRa, 2020-09-08:
Hello. 
I flashed KISS v1.9.2.0
Set Amperage = "Use direct connect current sensor"
It works OK - amps and mAh values change as expected.

I try to find manual for furling32 but could not find. It does look like there is an analog current out. Maybe ask Airbot (manufacturer) if there is a manual available?

hulandy, 2020-09-09:
I tried to connect the current output  to a betaflight FC, at the first time it didn't work either, but when I connect the positive to the voltage in of the osd it just worked. So, I think it must be a wiring issue, but I don't know how to solve it..

ShikOfTheRa, 2020-10-09:
Did you solve this?

closed 2020-10-21 by ShikOfTheRa