ShikOfTheRa/scarab-osd #621: RPW PID for INAV

Drone-pilot, 2020-10-30:
I know that this issue has already been raised, but it is very important for us! Please add support for displaying RPW PID values for in-flight configuration, as not all flight controllers have built-in OSD and INAV does not support configuring external OSDs on some controllers.INav developers don't want to implement this.

ShikOfTheRa, 2020-11-01:
Hello - I have not used this functionality - can you point me at a video which shows it in operation to see if it is possible to replicate it?

Drone-pilot, 2020-11-01:
Yes, of course, only the video is not mine, but from the RCSchoolmodels channel.  https://youtu.be/Xqtb2ns1k24

ShikOfTheRa, 2020-11-03:
OK I have this in development. It is working on the bench pending a test in a real quad and updating the GUI to support it.

When done, would you be able to confirm it's OK?
[image]

Drone-pilot, 2020-11-04:
Cool! I want to test it! How do I configure this in MWOSD-configurator 1.9.1.1?

ShikOfTheRa, 2020-11-04:
can you send me a mail - to:    spam <@> flypix.co.uk   and I'll reply from there with details
Its a new release and updated GUI needed to support it.

Drone-pilot, 2020-11-04:
OK, I sent you an email.

ShikOfTheRa, 2020-11-06:
OK it is done but will be available for 2.0 

I have emailed with the files if you want to use before it is released mainstream.

I will close this and mark as done - any issues just send me an email or re-open!

closed 2020-11-06 by ShikOfTheRa

Drone-pilot, 2020-11-11:
Hi! I tested version 2.0.0.0 of MWOSD and I couldn't configure the PID because there are several problems: 1. when loading settings from a file created in version 1.9.1.1, all the labels were not in their places, as well as the selected menu items of the displayed settings were changed; 2 after configuring OSD in 2.0.0.0 MWOSD and micro minimosd firmware in fpv glasses, all the labels are randomly shifted, although when reading in the program everything is displayed exactly. I attach printscreens, in printscreens 01 and 02 it is displayed as OSD was configured in MWOSD version 2.0.0.0, in printscreens 10 and 11 it is shown what I had depicted in FPV glasses. And attach the settings files in MWOSD-settings-2020-10-09T18-42-09.json settings for version 1.9.1.1, MWOSD-settings-2020-11-08T12-53-03.json settings for version 2.0.0.0. Photos with FPV glasses did not work, maybe I will do it in the evening.

Drone-pilot, 2020-11-18:
ShikOfTheRa, thank you for the work done!  Everything you've done is very cool!  But there are small bugs in the work, I wrote to you by e-mail about it.

ShikOfTheRa, 2020-11-20:
Sorry - emails hit spam folder...

I will reply