ShikOfTheRa/scarab-osd #622: Help garbage font of mwosd 1.9.2

abimaputra78, 2020-10-31:
Dear Shiki,

Presently, I used a paid version mwosd 1.9.2 for burning mwosd kiss gps osd. I used FTDI and minim micro osd like usuaI, tried several times... Eeprom clear,  re-flash, the osd font seemed... strange or there was some overlay of my old osd kiss firmware fonts and icon. How will the solution be? Pls assist and advice. I need to use mwosd for kiss gps. 

Pls assist how to set LQ Crossfire with channel 8=LQ. 80 warning, 70 critical warning

Thanks 
Ditto.

ShikOfTheRa, 2020-11-06:
Are you working following emails?

Support is from community here:
https://www.rcgroups.com/forums/showthread.php?3131701-MINIM-OSD-%C2%96-MWOSD-software-for-Ardupilot-iNAV-Betaflight-KISS-GPS-OSD-PX4

This site is intended for development issues.

ShikOfTheRa, 2020-11-20:
Hi Ditto - will close this as we got you working
CH8 resolved for next release.
KISS Crossfire and GHOST  LQ. Added to tracker for development. No guarantees, but will try...
They are available as RSSI values now if Crossfire etc to output that as an RSSI channel

closed 2020-11-20 by ShikOfTheRa