ShikOfTheRa/scarab-osd #624: Some ideas for GUI improvement.

RMFPV, 2020-11-11:
1. Font editor is brilliant. The only feature it's missing is moving the whole character inside the box. Sometimes it's necessary to change character position slightly.
2. It would be very helpful to store the current path separately for the settings file, telemetry player, and font directory.

error414, 2020-11-14:
Hi,

1) good idea, I'm going to add it to next version https://www.youtube.com/watch?v=7RGpgz4JuXo
2) I will try look at that

RMFPV, 2020-11-14:
Great! Thanks!
Video is private, though.

I would also suggest changing Aero in FW selection for something more common like Fixed Wing or Plane.

error414, 2020-11-14:
I has fixed video.

It's auto generated text from https://github.com/ShikOfTheRa/scarab-osd/releases/tag/1.9.2.0 I can add "translation" but it's low priority change.

ShikOfTheRa, 2020-11-20:
Agreed - plane is done ready for next release

ShikOfTheRa, 2020-11-20:
OK - I'll close as done ready for next release

closed 2020-11-20 by ShikOfTheRa