ShikOfTheRa/scarab-osd #625: single cell average voltage

corrado-c, 2020-11-15:
Hello!!!
i am using microminim osd in different constructions,
the thing I miss most is having the ability to read the SINGLE CELL AVERAGE VOLTAGE.
I often use direct connect voltage, without FC, GPS osd, for RC CAR so to read voltage and speed.
but also in the other configurations I think it is very convenient because I often change the type of batteries.

Maybe it is me who am not good at setting this function in cofigurator. :)

I hope I got the right branch, I don't know which mwosd is referring to.

thanks !!

corrado-c, 2020-11-15:
damn me ...
only now i saw that it can be done via arduino IDE ...
it took me a while to set it up but now it works.
if I understand correctly: VOLTAGE VIDEO becomes average cell voltage,
quite right?

RMFPV, 2020-11-16:
yes

ShikOfTheRa, 2020-11-20:
This is also available as an additional GUI configurable item from the next release. 
[image]

ShikOfTheRa, 2020-11-20:
OK - I'll close as done ready for next release

closed 2020-11-20 by ShikOfTheRa