ShikOfTheRa/scarab-osd #627: FC passthrough not working with Kiss

AngusCox, 2020-12-14:
Talking with Alexander Fedorov, one of Kiss developers, there could be a way to use Kiss FC as passthrough. The GPS configuration inside Kiss could be use to read and write the OSD. Setting the correct com speed I was able to receive 0 (ceros) from the OSD, but on the other hand I wasn't able to connect the GUI to the OSD using the passthrough feature.
Is there any possibility to check this issue and make it work for Kiss users? Not having to solder and resolder again each time we want to make an OSD change will be great.
Thanks

ShikOfTheRa, 2020-12-18:
Hi - if we send you new GUI would you be able to test?

AngusCox, 2020-12-18:
> Hi - if we send you new GUI would you be able to test?

Sure, I can test it without a problem.

ShikOfTheRa, 2020-12-21:
Not able to create a MAC version at present. Would you be able to test a PC version in any way?

AngusCox, 2020-12-21:
> Not able to create a MAC version at present. Would you be able to test a PC version in any way?

Yes, I can use a VM to install Windows and the GUI. But on the other hand I only have the licensed version for Mac. So I will not be able to test the write option with the PC version.

ShikOfTheRa, 2020-12-30:
Hi - can you send an email top "spam@flypix.co.uk" - will reply with gui and kiss  v2 file to test.

AngusCox, 2020-12-30:
> Hi - can you send an email top "[spam@flypix.co.uk](mailto:spam@flypix.co.uk)" - will reply with gui and kiss v2 file to test.

Done!

xaviergriffon, 2021-01-01:
Hi (Happy New Year),
This is a very interesting development, if you need another beta tester, don't hesitate to ask me.

AngusCox, 2021-01-03:
Using new Kiss v2.0.0.0 hex and same configurator version on OSX works perfectly.
[image]

Using same hex version but on Windows, I don't have the option to select Kiss from the connection options so it is not working for now.
[image]

ShikOfTheRa, 2021-01-04:
OK - it is good news with OSX! It is great work by GUI dev error414....

for Windows - my error, I uploaded wrong win version... updated and here if required
https://drive.google.com/drive/folders/1nB-tWqwoN-zKOJ7YhumGC0QTusEWCV6I

xaviergriffon, 2021-01-04:
I've tested all OS.
With KISS v2, it works great!
With KISS v1, it doesn't work. In the logs, we have the message "Set passthrought Fail" but the FC seems to be in "passthrought" mode because it is no longer possible to connect it to KISS GUI.

In the GUI, it would be good to put a message to remind the user to set the OSD as GPS under KISS GUI.

ShikOfTheRa, 2021-01-05:
Thanks Xavier. Yes only expect it to work with V2

Maybe a link to this?
https://github.com/ShikOfTheRa/scarab-osd/wiki/Passthrough

xaviergriffon, 2021-01-05:
Absolutely agree, the wiki is perfect !

closed 2021-03-01 by ShikOfTheRa