ShikOfTheRa/scarab-osd #633: multiwii2.4-baro alt displays 0

wleagle, 2021-03-29:
Maybe multiwii 2.4 is too old for maintainance, but I'm still using it on a pro mini board. I have a bmp180 barometer, and it is a fixedwing.

I have tried many time.  The alt can be correctly outputed in the multiwiiwingui configurator, however, it displays 0 (all of other informations are normal except the alt)on my microminimosd with the newest MWOSD firmware.

In multiwii 2.4, I have commented the "AUTOSENSEBARO"
In MWOSD gui, I have also chosen the "use baro altitude, mag heading" for FixedWing alt/heading (MSP FC only). And of course, the "Fixedwing" is uncommented in the section of installation type.

please help!

ShikOfTheRa, 2021-04-01:
Firstly, until armed, I am sure baro will always read zero
Secondly, in MWOSD GUI, FixedWing alt/heading (MSP FC only) = Use BARO altitude with/without mag
Thirdly - yes, need to slect fixed wing
I think autosense needs to be removed now it is configurable in GUI. I would disable both of those in config.h

I can try to test next week.

ShikOfTheRa, 2021-04-06:
OK, tested and it appears to work as intended. The emulator in the GUI is multiwii based so great for test verification

#1 - GPS altitude is "zeroed" at arm. So if 1500m above sea level, arm and fly to 1600m, display will show 100m
#2 - There are 2 altitude display options - altitude and gps altitude. Altittude will show whatever is defined as the source in the other tab - baro or GPS
#3 "GPS altitude"  element will show zero when disarmed.
#4 "Altitude" element will show GPS altitude relative to launch (zero if disarmed) or baro altitude
#5 baro altitude I think is always set to zero by the FC when disarmed if I recall with Multiwii

in example below with fixedwing alt set to Use BARO, I send test data with emulator. Baro 160m altitude is sent by fc. GPS alatitude was 1600m when armed, then increased to 2200, so shows GPS altitude of 600m above launch position. Just test data so BARO/GPS do not match.... 
[image]

ShikOfTheRa, 2021-05-07:
Closing as no update in over a month. Please re-open if needed

closed 2021-05-07 by ShikOfTheRa