ShikOfTheRa/scarab-osd #637: MicrominimOSD with Inav

MsnSazabi, 2021-05-07:
Hello,

I'm having issue running OSD with inav. OSD fw ver 1.9.2, target inav, inav fw 2.6.1. Inav port sends data in msp at 115200. Port is working, checked by reading port output. What am I doing wrong?

ShikOfTheRa, 2021-05-07:
tx/rx wires crossed?

MsnSazabi, 2021-05-07:
> tx/rx wires crossed?

nope, wired correctly.

ShikOfTheRa, 2021-05-07:
Actually I have exact setup on one of my tricopters so it should work 

The GUI has a built in emulator with is iNav compatible - if you are able to hook up PC and use that whilt still seeing the OSD output this can be used to rule out OSD.

Usual issues are:
Not uploading correct firmware to OSD 
OSD TX connected to FC TX, OSD RX to FC RX instead of OSD TX to FC RX and OSD RX to FC TX
Baud rates not matching either side. If using iNav, use 115k
Not selecting MSP on the iNAV serial port in the iNav GUI ** easy one to miss. Catches me all the time.

Can sometimes get odd results if using Serial 1 on FC for MSP when connected to USB at same time.

Might be something in here that will help:
https://github.com/ShikOfTheRa/scarab-osd/wiki/Troubleshooting-Connectivity

MsnSazabi, 2021-05-07:
> Actually I have exact setup on one of my tricopters so it should work
>
> The GUI has a built in emulator with is iNav compatible - if you are able to hook up PC and use that whilt still seeing the OSD [...]

Thanks. It works with msp emulation. I guess, it's something with inav than.

closed 2021-05-07 by MsnSazabi

ShikOfTheRa, 2021-05-07:
OK - maybe it helps a little in narrowing the fault down.

Also hardware considerations - not using an inverted serial port or possibly serial 1 issue like mentioned.

Good luck!

MsnSazabi, 2021-05-09:
> OK - maybe it helps a little in narrowing the fault down.
> 
> Also hardware considerations - not using an inverted serial port or possibly serial 1 issue like mentioned.
> 
> Good luck!

Tanks. Btw, how to change MWOSD port speed? Just found out inav softserial is limited to 19200. And my osd is on softserial.

ShikOfTheRa, 2021-05-09:
OK. That for sure would explain.

That requires a custom build. Follow this guide:
Settings you need to amend (remove leading // in config.h) are:
#define iNAV
#define BAUDRATE 19200
and then change GUI speed to 19200 to be able to connect

https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide

MsnSazabi, 2021-05-09:
> OK. That for sure would explain.
>
> That requires a custom build. Follow this guide:
> Settings you need to amend (remove leading // in config.h) are:
> #define iNAV
> #define BAUDRATE 19200
> and then change [...]

Thanks!