ShikOfTheRa/scarab-osd #638: Mavlink OSD - some modes has wrong names

RMFPV, 2021-05-07:
TakeOff mode is displayed as ARM
AUTO displayed as WAYP which is not common

ShikOfTheRa, 2021-05-07:
Can you advise:
which type / version of FC firmware? 
also aircraft type 

And the MWOSD target

Normally the modes rarely change between versions, need exact info to replicate

ShikOfTheRa, 2021-05-07:
Just had a look in the code

I have looked at Ardupilot code and:
1 Renamed WAYP to AUTO for APM/PX4 users
2 Added mode for Take off  TKOF (fixed wing only)

This will be in next release. I hope in next few weeks.
Take off was not in documentation before. I also checked minim-osd-extra and it does not appear in there either. Not sure if its a new mode, but adding is not an issue.

These are modes listed in APM.:

        STABILIZE =     0,  // manual airframe angle with manual throttle
        ACRO =          1,  // manual body-frame angular rate with manual throttle
        ALT_HOLD =      2,  // manual airframe angle with automatic throttle
        AUTO =          3,  // fully automatic waypoint control using mission commands
        GUIDED =        4,  // fully automatic fly to coordinate or fly at velocity/direction using GCS immediate commands
        LOITER =        5,  // automatic horizontal acceleration with automatic throttle
        RTL =           6,  // automatic return to launching point
        CIRCLE =        7,  // automatic circular flight with automatic throttle
        LAND =          9,  // automatic landing with horizontal position control
        DRIFT =        11,  // semi-autonomous position, yaw and throttle control
        SPORT =        13,  // manual earth-frame angular rate control with manual throttle
        FLIP =         14,  // automatically flip the vehicle on the roll axis
        AUTOTUNE =     15,  // automatically tune the vehicle's roll and pitch gains
        POSHOLD =      16,  // automatic position hold with manual override, with automatic throttle
        BRAKE =        17,  // full-brake using inertial/GPS system, no pilot input
        THROW =        18,  // throw to launch mode using inertial/GPS system, no pilot input
        AVOID_ADSB =   19,  // automatic avoidance of obstacles in the macro scale - e.g. full-sized aircraft
        GUIDED_NOGPS = 20,  // guided mode but only accepts attitude and altitude
        SMART_RTL =    21,  // SMART_RTL returns to home by retracing its steps
        FLOWHOLD  =    22,  // FLOWHOLD holds position with optical flow without rangefinder
        FOLLOW    =    23,  // follow attempts to follow another vehicle or ground station

        ZIGZAG    =    24,  // ZIGZAG mode is able to fly in a zigzag manner with predefined point A and point B
        SYSTEMID  =    25,  // System ID mode produces automated system identification signals in the controllers
        AUTOROTATE =   26,  // Autonomous autorotation

and plane:
        MANUAL        = 0,
        CIRCLE        = 1,
        STABILIZE     = 2,
        TRAINING      = 3,
        ACRO          = 4,
        FLY_BY_WIRE_A = 5,
        FLY_BY_WIRE_B = 6,
        CRUISE        = 7,
        AUTOTUNE      = 8,
        AUTO          = 10,
        RTL           = 11,
        LOITER        = 12,
        TAKEOFF       = 13,
        AVOID_ADSB    = 14,
        GUIDED        = 15,
        INITIALISING  = 16,
        QSTABILIZE    = 17,
        QHOVER        = 18,
        QLOITER       = 19,
        QLAND         = 20,
        QRTL          = 21,
        QAUTOTUNE     = 22,
        QACRO         = 23,
        THERMAL       = 24,

ShikOfTheRa, 2021-05-07:
And no need for all the version info - I could figure it out from the code.

Thanks for pointing out. Every bit is an improvement.

RMFPV, 2021-05-07:
Takeoff  is quite a new mode in fact from v4.0 only for fixedwing. It was as a mission command previously.
Please note, that in flight it work as LOITER mode if used.

Thank you for quick update!

ShikOfTheRa, 2021-05-09:
Do you have a tlog which includes takeoff so I can test?

Cheers

RMFPV, 2021-05-09:
Sorry, I don't use MP in flight, so I don't have it. I will try to record next time.

ShikOfTheRa, 2021-05-10:
OK - no need. Probably easier to wait to test code in a few days. It should be OK... I think my tlogs are all older so do not have it. I might be able to find a newer one..

ShikOfTheRa, 2021-05-12:
OK to close?

RMFPV, 2021-05-12:
Yes, tested - all good!
Thank you!

closed 2021-05-12 by RMFPV