ShikOfTheRa/scarab-osd #643: large battery capacity

umeloman, 2021-09-06:
Hello! Recently I've got new issue, I think it's in OSD. On large drone battery capacity became to negative digit after 32750mAh (then it goes to -32750mAh and start going to 0). Looks like it is some max point of memory or some. Maybe it is better to change mAh to Ah for example  after 10000mAh and then show it like 10.21Ah, 32,76Ah or so? MicroMinimOSD + Ardupilot. Thank you!

umeloman, 2021-09-06:
Thanks for quick reply! Yes, I was first with question about 12S LiPo battery voltage and now capacity is around 64Ah. It will be fine with 1 digit after point, no problem. I'm glad it is possible at least and you are healthy and online! Thank you, good luck!

ShikOfTheRa, 2021-09-07:
BTW - can you share a bit of info about your aircraft? 12s / 64AH sounds very interesting!!

umeloman, 2021-09-07:
It is BD625 25KG (MTOW up to 50kg) Agricultural drone based on Hobbywing X9
propulsion system with 34" props
https://ae01.alicdn.com/kf/H8915c91ab39749afadfa7cfc40dbb5b4c.jpg

ShikOfTheRa, 2021-09-07:
OMG. It is huge. 
I have large tricopter that is 1m between motors and 15 inch props and is the biggest have ever seen

umeloman, 2022-02-11:
Hello! I hope you are fine! This question is still actual for me. Thank you!

ShikOfTheRa, 2022-02-14:
Hi - sorry for delay - it is not forgotten - I have just been busy with day job and finishing off another development project.

MAybe 4 weeks I will be free to complete

umeloman, 2022-06-25:
Hi! How are you? This thing is still actual for me. Now I'm using ver 2.0 pre release, also added GPS Alt and Baro Alt and both are not correct when GPS signal is lost

ShikOfTheRa, 2022-12-17:
Hello, is this still required. Today I found my development unit after 1 year in storage so will be completing a few requests
If yes, does the amp draw display correctly? And what is the maximum amp draw approx of your drone

ShikOfTheRa, 2022-12-18:
Hi - this is done. 
It is not available in pre-compiled binaries that can flash from GUI until next release. If this is an issue, or not sure how to build your own using arduino, please re-open.

https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide

closed 2022-12-18 by ShikOfTheRa

umeloman, 2023-12-26:
Thank you! Now it is mismatch with the simbols mAh to Ah. I can draw Ah instead of mAh, but up to 9999mAh there is no  "m" letter, like meters in distance ot altitude values. Itneresting challenge for me =)

ShikOfTheRa, 2023-12-28:
Hi - here is a temporary solution. It is not available in pre-compiled binaries that can flash from GUI so it will require:

1 Custom installation ( flash using arduino )
https://github.com/ShikOfTheRa/scarab-osd/wiki/Custom-installation-guide

2 Load fonts (Edit, load font from file, and select file from fonts folder in zip file below)

https://github.com/ShikOfTheRa/scarab-osd/archive/refs/heads/Battery-display-AH.zip

I am not planning any more updates for MWOSD. It is near end of life with very little capacity and also the number of users have reduced as they migrate to newer platforms. MWOSD will probably remain static now for many years.