ShikOfTheRa/scarab-osd #645: miniosd

ricardovalmeida, 2021-11-17:
hi i am having difficulties in making a miniosd card work. I want to make a project and I would like to know if anyone has the firmware that I can write a simple text with this osd board, I'm learning the program with arduino, I'm using this code but it's not working.
[image]
[HelloWorld.txt](https://github.com/ShikOfTheRa/scarab-osd/files/7558394/HelloWorld.txt)

ShikOfTheRa, 2021-11-24:
Maybe try this:

https://github.com/ShikOfTheRa/MAX7456

ricardovalmeida, 2021-12-02:
thanks, ShikOfTheRa, I'll run some tests

closed 2022-01-28 by ShikOfTheRa

ricardovalmeida, 2022-02-28:
hi ShikOfTheRa
you helped me a lot with the osd, I'm already learning to write on the tv image. I need to make a counter for people who come to test for covid, I'm going to send a 50us for the osd to count, what would be the best variable for a counter.
I accidentally recorded another character font in the MAX7456 and I am not able to write it correctly, you have the original table that comes in the ci so that I can record the max7456 again.
Thanks