ShikOfTheRa/scarab-osd #647: OSD layout reorganized after updating MWOSD from 1.9 to 2.0.

wcfung, 2022-06-22:
The re-organization of display layout happened on following conditions :--
1)  happen on both Minim OSD AND Micro Minim OSD with kv mod, 
2)  happen ONLY when transmitter is turned ON. If transmitter is not turned ON after powering up the vehicle and video system, the display maintains the same layout as that when I initially configured it.
3)  happen only on Arduplane FW, not on Arducopter FW. Both FW are from Ardupilot.

Re-organization of layout means parameters at the bottom of screen shift one row down, with some param relocated to the top of screen, those initially at bottom left shift to top right.

FW of my Arduplane is 4.1.7, and that of Arducopter is 4.1.2.

Appreciate any help, as I have no knowledge on coding.

Thank you.
Frankie

ShikOfTheRa, 2022-06-27:
Hi Frankie - try setting in other tab / Video signal type. I thin this might be what you need.

In default AUTO, to have same screen in NTSC/PAL, it inserts extra blank lines. This can be manually changed by selecting PAL and arranging items how you would like.

wcfung, 2022-06-28:
Hi,
Thanks for the reply. I changed the video signal type to PAL, save the change and tried again, nothing changed.  I tried NTSC, still with same result. This shifting of lines happened only after the Tx is turned ON, so seems has nothing to do with the type of video signal, because before I turned ON the Tx, the vehicle, and therefore the video Tx are both powered up, and at that point in time, there is no shifting in lines. Pls see attached photos before and after Tx is turned ON.
[image]
[image]

Another point worth repeating again...... this shifting does not happen in Arducopter firmware. 
Any other hint you can think of ? Appreciate your help.
Frankie

wcfung, 2022-06-29:
> That looks like different screen layouts to me rather than items shifting
> around.
>
> I would connect via GUI and double check the three tabs - LOW/MID/HIGH on
> the preview.

Hi,

You are right, it is not shifting of lines. My OSD Screen switch was set to
"USe RC channel" instead of "Use FC OSD
Switch".
It works fine now.

Thank you for yr help. Cheers
Frankie

wcfung, 2022-07-12:
Hi,

Could you please help me again on the MWOSD and my Minim OSD.

I recently flashed the Minim OSD with MWOSD 2.0,, as follows :--

Version : 2.0
Device :  I tried both microminim and minim
Variant : Ardupilot
Type : aero

After loading online firmware .HEX and upload font, the display in the
monitor did not update.
The FC is Revolution. Camera image display is okay, just no updating in all
param except the timer.

Appreciate if you can suggest any possible reasons.

Regards
Frankie

ShikOfTheRa, 2022-07-12:
I assume you are using ardupilot still.

Here is setup info. Mainly serial port for Ardupilot must be set correct. You can also try "autoconfigure" in mavlink tab on MWOSD GUI
https://github.com/ShikOfTheRa/scarab-osd/wiki/MAVLINK-installation

wcfung, 2022-10-11:
Thanks for the very useful link which I have bookmarked it for future
reference.

I got the OSD update working as I did not check the serial port which set
to other value in previous application.

Cheers
Frankie

closed 2022-10-27 by ShikOfTheRa