ShikOfTheRa/scarab-osd #662: Is it possible to inject custom information from the FC into OSD?

skol101, 2024-03-26:


AKAMEDIASYSTEM, 2024-08-05:
I'm interested in this too - specifically, I have Lua scripts that send custom MAVLink1 STATUSTEXT (id 253) messages that I'd like to show up onscreen in a manner similar to the way they do on Mission Planner.
Any chance this functionality is available, and I'm just missing how to enable it in the GUI?
Thanks for making this software, btw, it's very helpful and definitely worth the two pounds!

error414, 2024-08-06:
it's not possible, tracker sends only 32 bits :(

AKAMEDIASYSTEM, 2024-08-12:
Hi, I actually got this working just fine without needing to alter the scarab-osd firmware - my issue was most likely just due to an upstream issue sending mavlink statustext messages on the right `send_chan`, which I figured out by trial and error.
Now I am able to send custom messages from ardupilot lua script to the osd display. Sorry to have maybe hijacked/confused this github issue!

closed 2025-04-15 by ShikOfTheRa